markdegrootnl/unifi-protect-arm64 #2: Arm64 Protect crashes in the background

DSteinmann, 2021-04-19:
Hi,
Thank you for creating this container! I can't seem to get this container running reliably. I did get it to start the first time but as soon as I try to load the protect application, I always get a 500error in the frontend and the application won't load at all. I did get it to load for a few seconds but it couldn't connect to the local nvm (Trying to connect to your nvm). 

It looks like there are a bunch of TLS certification errors in the log files within the container but I can't seem to resolve them. I've already tried a reverse proxy with a custom certificate but to no avail... 
Do you have any tips or similar experiences with this error? 

I'm trying to run it on a Pi 3+ with ubuntu 20.04 x64

Thanks a lot!

ozett, 2021-12-23:
> @gtrak You can follow the steps in the [Dockerfile](https://github.com/markdegrootnl/dockerfiles/blob/master/unifi-protect-arm64/Dockerfile)

link is gone.. 
what is the hint from there?

(as i search also for a solution of the error on the end of setting up the protect-assistant..)

markdegrootnl, 2021-12-23:
Here is the correct link: [Dockerfile](https://github.com/markdegrootnl/unifi-protect-arm64/blob/master/Dockerfile )

@ozett what's your question?

ozett, 2021-12-23:
> @ozett what's your question?

trying to get your arm docker-image running, but ran into this error shortly before success
https://github.com/markdegrootnl/unifi-protect-arm64/issues/6#issuecomment-1000139016
now looking for answers how to overcome this unspecific error.
found that thread and hoped to find answer in the dockerfile. but link was gone.

maybe the error has a different cause. i can bash into your dockercontainer. 
looks all like a normal linux, but i have no clue what to check to fix that error..

any hints appreciated.. 👍

gtrak, 2021-12-23:
I'm not sure what is the cause, I ran it on an rpi 4 4GB. I'm no longer trying to use it that way. It seemed it was set up correctly, just the server would crash repeatedly. It would still serve some pages. My guess is just not enough RAM.

markdegrootnl, 2021-12-23:
Maybe limit the allowed memory to the container by adding `-m 2048m` to the run command.

markdegrootnl, 2022-01-03:
Please update the docker image to the latest version 1.20.3

closed 2022-01-03 by markdegrootnl