markdegrootnl/unifi-protect-arm64 #3: Unstable container

Thijxx, 2021-08-03:
The container generates 60% CPU load with 0 camera's connected. 24/7.
The container is unstoppable, there is absolutely no way to stop it. Not with docker, not with kill -9, just no way.
Trying to brute force stop the container halts the system, then it needs a power cycle.

markdegrootnl, 2021-08-04:
"Unstoppable", it sounds like a good thing...

I personally don't experience this behaviour. Which container version are you running? Are there any logs that might provide more insight?

9dc, 2022-01-03:
with the current version of this container, i didnt had the problem again. i think its fixed.

markdegrootnl, 2022-01-03:
@Thijxx How's your experience running the latest version?

Thijxx, 2022-01-03:
@markdegrootnl I did not continue with the camera's but can try again as a nice start of the new year :D

Thijxx, 2022-01-03:
The extreme CPU usage seems to be fixed. That is nice! Running it is still hard, using docker-compose it finally worked but the the setup failed with the completely useless message: "An unexpected error occured during setup." But that's just Ubiquiti quality software :D

closed 2022-01-03 by Thijxx