markdegrootnl/unifi-protect-arm64 #5: Container fails to start

9dc, 2021-09-30:
Every time i want to start the container it gives me this error in the Log:

standard_init_linux.go:207: exec user process caused "exec format error"

`docker run -d --name unifi-protect-arm64  \
    --privileged \
    --tmpfs /run \
    --tmpfs /run/lock \
    --tmpfs /tmp \
    -v /sys/fs/cgroup:/sys/fs/cgroup:ro \
    -v /home/protect-storage/srv:/srv \
    -v /home/protect-storage/data:/data \
    --network host \
    markdegroot/unifi-protect-arm64:latest`

Any idea?
Running on a fresh Raspberry Pi 4 8 GB
200 GB SD Card

markdegrootnl, 2021-09-30:
Hi 9dc, 

Please run the following command on your Pi and post the output: `uname -a`. 

I think you're running a 32 bits OS on the Raspberry Pi. You need a arm64 operating system to run the container.

9dc, 2021-09-30:
thanks now it works! i had 32 bit

closed 2021-09-30 by 9dc