markdegrootnl/unifi-protect-arm64 #6: UNVR Setup Error: An unexpected error occured during setup.

MastaP, 2021-10-10:
Hi,

Started the container using `docker run` command. In web UI, after pressing Finish in the setup process I get the error. 
First, I tried restarting the container and setting folder permissions to 777, did not help.
Inside the container `more /var/log/postgresql/postgresql-9.6-main.log` had:
```
2021-10-10 14:52:20.424 UTC [330] postgres@ulp-go FATAL:  database "ulp-go" does not exist
2021-10-10 14:52:21.473 UTC [353] postgres@ulp-go-syslog FATAL:  database "ulp-go-syslog" does not exist
2021-10-10 15:33:34.099 UTC [82] FATAL:  data directory "/data/postgresql/9.6/main/data" has group or world access
2021-10-10 15:33:34.099 UTC [82] DETAIL:  Permissions should be u=rwx (0700).
```
So I've set the permissions to 700, the error above went away.

Then, looking into `/srv/unifi-protect/logs/errors.log`:
```
2021-10-10T15:48:58.410Z - error: Cannot connect to database. connect ENOENT /var/run/postgresql/.s.PGSQL.5433
2021-10-10T15:48:58.420Z - error:  Error: connect ENOENT /var/run/postgresql/.s.PGSQL.5433
    at PipeConnectWrap.afterConnect [as oncomplete] (net.js:1191:14)
2021-10-10T15:49:33.274Z - error: DEVICE NOT REGISTERED unregistered-hardware
```

Any ideas?

ozett, 2021-12-23:
similar error after starting docker on my jetson-nano
enough space, but similar errors.
any hint or solution how to overcame this?

[image]

markdegrootnl, 2021-12-23:
Which version of the Docker container are you running?

MastaP, 2021-12-23:
I used 1.15.0

ozett, 2021-12-23:
Docker version 20.10.7, build 20.10.7-0ubuntu5~18.04.3

fresh jetson nano-os:
[image]

ozett, 2021-12-23:
must this be fixed?

[image]

ozett, 2021-12-23:
exchanged the jetson-storage (was an nfs-mount for 100 GB free) with a usb-ssd on ext4.
no luck.
also not used the :latest docker-image
no luck.

is this meant to be run without error on the docker-command from the readme.md?
i think so (like others) but now a littel clueless how to get it running..

maybe some mdadm/ub-storage defaults or mappings are missing 
to assemble the /md3 raid from the inital setup?

[image]

ozett, 2021-12-23:
maybe its the firmware 2.2.7 itself? update [to 2.2.12](https://community.ui.com/releases/UniFi-OS-Network-Video-Recorders-2-2-12/419e87b7-0774-4a57-b26c-d1a0f3400aab?page=2) ?

```
root@UNVR:/# cat /usr/lib/version
UNVR4.al324.v2.2.7.2aa2917.210824.1401
root@UNVR:/# cat /proc/md
```

ozett, 2021-12-23:
i cleared existing images and pulled the latest.
to make shure its version 1.20.1 i specified this for the docker-run, but it pulled (anohter?) again..
will report later the result...

```
docker rm $(docker ps -a -q)
docker rmi $(docker images -q)
docker pull markdegroot/unifi-protect-arm64:latest
```
[image]

ozett, 2021-12-23:
did not help

any other things to check ?
-- inside the container running the asisstant?
must there be (mapped) raid-disk inside? which?

can i stay with this? raid10 also thows the same error..

[image]

markdegrootnl, 2021-12-23:
You just select the "One Disk" option.

Could you check the logs inside the container in `/data/unifi-core`

ozett, 2021-12-23:
i went through some of the logs... but did not catch a hint...
the ub-storage service complaints about missing /dev/md3

i assume that the assistant does not assemble the raid .
maybe because of faulty docker-mappings for disk-path into the containter?

or something specific to look for?

WEIRD cleaning of images as i wanted to see if that also happening with image tagged 1.19.0...
will report in a minute...

[image]

ozett, 2021-12-23:
ver 1.19.0 also not running on the jetson-nano arm...
some specific differences to the raspberry? naming of the disc?

[image]

ozett, 2021-12-23:
> You just select the "One Disk" option.
> 
> Could you check the logs inside the container in `/data/unifi-core`

dont know whats relevant, but  in ```system.log``` (need to fix this?)
```

2021-12-23T14:02:09.787Z - error: error retrieving setup device id Error: EACCES: permission denied, mkdir '/persistent/unifi-core'
2021-12-23T14:02:21.187Z - error: error retrieving setup device id Error: EACCES: permission denied, mkdir '/persistent/unifi-core'
2021-12-23T14:02:31.333Z - error: error retrieving setup device id Error: EACCES: permission denied, mkdir '/persistent/unifi-core'
2021-12-23T14:02:31.551Z - error: error retrieving setup device id Error: EACCES: permission denied, mkdir '/persistent/unifi-core'
2021-12-23T14:02:33.768Z - error: error retrieving setup device id Error: EACCES: permission denied, mkdir '/persistent/unifi-core
```

ozett, 2021-12-23:
The assistand never finishes, and i see port-errors in the log,
but thats the status without getting finish:

[image]

ozett, 2021-12-23:
any idea what to check for the building of the raid from assistant??
found some info for that, but no clue how to assemble it by hand into the docker containter...
https://community.ui.com/questions/Trouble-adding-NVR-to-full-Unifi-network-UniFi-Protect-NVR-Setup-Error/51d9f514-4da3-42cd-ba8a-835699eb2fc4#answer/a4ca0c1f-2fb7-4327-98b7-4356f839fce5

ozett, 2021-12-23:
this also did not help:
[image]

i made some empty files, mounted them as loop-devices and mapped them into the container as disks
```
dd if=/dev/zero of=raid-2 bs=1M count=500
dd if=/dev/zero of=raid-3 bs=1M count=500

losetup /dev/loop2 raid-2
losetup /dev/loop3 raid-3
```

```
docker run --name unifi-protect-arm64  \
    --privileged \
    --tmpfs /run \
    --tmpfs /run/lock \
    --tmpfs /tmp \
     --shm-size=64m \
    -v /sys/fs/cgroup:/sys/fs/cgroup:ro \
    -v /media/olaf/ff3594e6-3f3f-4133-98c3-1ad49278c650/srv:/srv \
    -v /media/olaf/ff3594e6-3f3f-4133-98c3-1ad49278c650/data:/data \
    -v /dev/loop2:/dev/sda2 \
    -v /dev/loop3:/dev/sda3 \
    --network host \
    -m 2048m \
    markdegroot/unifi-protect-arm64:latest
```

still error at the end of the assistant...
any clues?

ozett, 2021-12-23:
fail, fail, fail...
[image]

ozett, 2021-12-23:
are there files missing?
[image]

ozett, 2021-12-23:
fail with docker image v1.19.0 & 1.18.0
succes with docker v 1.17.3

[image]

ozett, 2021-12-23:
it runs unstable, 
but all i wanted was to activat my g3 instant.
that somehow was succes with the android app on your docker-image v1.17.3....
thanks a lot.
(the whole thing seems to be unusable for production. somehow only for lab-experiments..)

[image]

ozett, 2021-12-23:
the app adopted the cam, made an error.
cannot see how to switch the cam now to stand-alone mode (like the g3) or if its possible at all..
[image]

Thijxx, 2022-01-03:
Having the same problem here..

Lots of these in **errors.log** in **unifi-core**: `stderr: /bin/sh: 1: ustorage: not found`
Also: `mdadm: cannot open /dev/md3: No such file or directory`

**trace.log**
```
2022-01-03T09:20:00.890Z - warn: Error during metric send
2022-01-03T09:20:00.891Z - warn:  FetchError: request to https://trace.svc.ui.com/traces failed, reason: unable to get local issuer certificate
    at ClientRequest.<anonymous> (/usr/share/unifi-core/app/webpack:/node_modules/node-fetch/lib/index.js:1483:11)
    at ClientRequest.emit (events.js:315:20)
    at TLSSocket.socketErrorListener (_http_client.js:426:9)
    at TLSSocket.emit (events.js:315:20)
    at emitErrorNT (internal/streams/destroy.js:92:8)
    at emitErrorAndCloseNT (internal/streams/destroy.js:60:3)
    at processTicksAndRejections (internal/process/task_queues.js:84:21)
```

Also loads of "sudo password needed" in the logs.

markdegrootnl, 2022-01-03:
The errors are fine. That's just due to running on own hardware.
Please update the docker image to the latest version 1.20.3

closed 2022-01-03 by markdegrootnl