markdegrootnl/unifi-protect-arm64 #7: Failed to Create /init.scope control group

craig395, 2021-12-18:
Hi, 

I am receiving the following output on the HDMI output when trying to run the container, even though I am running the command while connected via ssh. Nothing obvious happens on the ssh end.

> Failed to create /init.scope control group: read-Only file system
> Failed to allocate manager object: Read-Only file system
> Freezing execution

I have very little experience with Docker but looking into this I have so far seen it may be related to some compatibility between docker and systemd 
info: https://serverfault.com/questions/1053187/systemd-fails-to-run-in-a-docker-container-when-using-cgroupv2-cgroupns-priva

My system:
OS: Ubuntu 64bit server edition
hardware: Pi 3
Other: Connected to tv via HDMI, mostly using remotely via ssh. connected via Ethernet to network.

Any help is appreciated, I will update if I find out more or get to a solution.

Sputnick85, 2022-04-12:
ok so had the same problem. You can solve this by creating the following json config at the docker host

>/etc/docker/daemon.json

`{ 
"exec-opts": ["native.cgroupdriver=systemd"],
  "features": { "buildkit": true },
  "experimental": true,
  "cgroup-parent": "docker.slice"
}`

but that's just half the work. after this i had to edit the unit files of unifi-core and unifi-protect and add the following line under _[service]_ section

> /etc/systemd/system/multi-user.target.wants/unifi-protect.service
> /etc/systemd/system/multi-user.target.wants/unifi-core.service

`TasksMax=infinity`

at the end i had the "hostname error" [mentioned in this post](https://github.com/markdegrootnl/unifi-protect-arm64/issues/1#issuecomment-1012098965)  so you should edit the change_hostname function in _/sbin/ubnt-systool_ as well

craig395, 2022-06-07:
Thanks!
Adding the "TasksMax=infinity" text to the two files and using the command in the post linked in your message got it working!.

Steps I did all together.

After updating everything and re-downloading the docker image I tried these steps.
First, when I checked the "/etc/docker/daemon.json" file it was already set to the text you provided as below. 

> { "exec-opts": ["native.cgroupdriver=systemd"], "features": { "buildkit": true }, "experimental": true, "cgroup-parent": "docker.slice" }

I then used the command in the [comment you linked ](https://github.com/markdegrootnl/unifi-protect-arm64/issues/1#issuecomment-1012098965) to make the docker container instead of the one on the Github page.

> docker run -d --name unifi-protect-arm64  \
>     --privileged \
>     --cap-add SYS_ADMIN \
>     --security-opt seccomp=unconfined \
>     --tmpfs /var/run \
>     --tmpfs /var/run/lock \
>     --tmpfs /tmp \
>     -v /storage/srv:/srv \
>     -v /storage/data:/data \
>     -v /storage/persistent:/persistent \
>     --network host \
>     --cgroup-parent=docker.slice \
>     --cgroupns private \
>     markdegroot/unifi-protect-arm64:latest

I was then getting the fork error mentioned there.
I updated the file as mentioned in that comment by vPotusFlotus, but this did not remove the fork error.
At this point, I realized the two files that needed "TasksMax=infinity" to be added were on the container and not the host machine.
I then installed VIM on the docker container. 

> docker exec [container-ID] sudo apt-get -y install vim

Next, I ran the below in order to input commands to it (I'm quite new to Docker)

> Docker exec -it [Container-ID] bash

I then used VIM to edit the files. Making sure to use ESC -> type ":wq" -> Enter key in order to save. (nano that I normally use didn't seem to work for me "Could not open terminal")

> VIM /etc/systemd/system/multi-user.target.wants/unifi-protect.service

and

> VIM /etc/systemd/system/multi-user.target.wants/unifi-core.service

At this point I could load the webpage. As both you and vPotusFlotus mentioned about the hosts file I edited the file "/sbin/ubnt-systool" mentioned in the comment.

After all this I went through the wizard and had it running. Although I think I may need to get a newer PI as it seems to struggle sometimes, especially after it does any update. 1GB RAM may be a little low.

Many thanks for the help.

markdegrootnl, 2022-06-08:
This problem occurs when your host-os uses cgroups2 exclusively and no cgroup v1. The protect container wants to start systemd inside the container but this is not possible with just cgroups2.
To make your host-os use cgroup2 and 1 together run **on the host os**:
```
echo 'GRUB_CMDLINE_LINUX=systemd.unified_cgroup_hierarchy=false' > /etc/default/grub.d/cgroup.cfg
update-grub
```
and restart.

craig395, 2022-06-08:
Thanks for the info, I have run that command but the update-grub is giving the error below. 
[image]
I have restarted after this so will see if it has an effect when it struggles. 

Most of the time I look at the console after its been running for a while I have messages on there like 

> Out of memory: Killed Process.......

and

> echo 0 > /proc/sys/kernal/hung_task_timeout_secs disables this message
>INFO: task kworker blocker for more than .... secs 
>tainted G             C

The out of memory makes me assure it's just low RAM..... 
Otherwise its been running ok

closed 2022-06-10 by markdegrootnl

gurkburk76, 2023-05-11:
So this is getting realllly annoing, i'm trying to get it working on a rasp  3b+ and tried all of the stuff about but it still fails to propperly start the container. running raspberrian (debian bullseye) and this is what pops up in journalctl -f
May 11 16:49:08 zigbee systemd[1]: Failed to create /init.scope control group: No such file or directory
May 11 16:49:08 zigbee systemd[1]: Failed to allocate manager object: No such file or directory
May 11 16:49:08 zigbee systemd[1]: Freezing execution.

I did try what mark wrote above and i', guessing that for my dist this is actually in /boot/cmdline.txt and it now looks like this:

console=serial0,115200 console=tty1 root=PARTUUID=a1201b44-02 rootfstype=ext4 fsck.repair=yes rootwait systemd.unified_cgroup_hierarchy=0 systemd.unified_cgroup_hierarchy=falseroot

Help much appreciated :)