markdegrootnl/unifi-protect-arm64 #11: Container starts, Protect doesn't

duncan999007, 2022-02-27:
Raspberry Pi 3 with Ubuntu Server 20.04

When I run the command listed in Readme, it pulls and starts a container that shows "/lib/systemd/systemd" as the command. Connection is refused if I check in a browser, unifi protect app doesn't find anything on the network, and a port scan shows no ports open. I've disabled firewall to be sure. 

Nothing is listed if I run "docker logs" and "attach" just seems to hang the console. 

I'm relatively new at docker so there may be more I can troubleshoot; just let me know. 

I also looked at building it myself. Got as far as pulling all the .deb files from the firmware when I found it doesn't have a /var/lib/version file. I tried to install unifi-protect on the machine itself and even installing every .deb from the firmware, there was always an unmet dependency. 

Any help is appreciated

xF4m3, 2022-02-28:
I got the same issue, I also tried installing it using the commands in the docker file on an debian 9 installation.
I got as far as having no dependency issues, but currently the services unifi-protect and unifi-core do not run, core is missing an settings.yaml and the other one seems like my node might be to new since it throws me an deprecation error.

But i cant get an older version of Node12 as an deb file, and the 12.18.0 that i got is to old (needs 12.18.1 or newer)

I think the docker is also broken, a docker container should normally work out of the box.

9dc, 2022-03-05:
for me everything was working perfect - yesterday i did "do-release-upgrade" and now its broken. 
the container is up and running but when i got to the webinterface i get "connection refused"

xF4m3, 2022-03-06:
@9dc Which OS were you on before? Ubuntu 18.04 on arm64? I assume you are also using a raspberry of some sort?

Afaik, the host OS should not affect the Containers, but if you say so, it might be worth a try for me to test this with the older OS.

9dc, 2022-03-06:
@xF4m3 
I thought so. Yes im Running a Pi 4 (8 GB RAM) 
I really dont know which Version i had before. I installed the pi about 6 months ago and just installed it with the pi imager stable arm64 build. Might be 18.04

xF4m3, 2022-03-06:
@9dc well 20.04 was definetly out 6 months ago, but you could still have installed 18.04, because from 20.04 you should not be able to do-release upgrade. So i would guess 18.04. But what you could try is rebuilding the container, I dont know how but a friend of mine said that after you upgrade to a newer OS version this could sometimes be needed. But I don't know if thats true or not, but it could be possible. I will try a VM with 18.04 and see if i can get the Container working in there.

markdegrootnl, 2022-03-07:
The Host OS should not have any influence if you're running the Docker container. 

You should not run any apt updates inside the docker container, this wil break it for sure.

If the container starts and unifi-core is running but unifi-protect is not, make sure you have enough storage space (100GB+) available for your /srv and /data docker volumes.

When building the docker container yourself, you'll need a couple of deb files and the version file from a UNVR firmware file. Documentation is in the `[put-deb-files-here](https://github.com/markdegrootnl/unifi-protect-arm64/tree/master/put-deb-files-here)` and `[put-version-file-here](https://github.com/markdegrootnl/unifi-protect-arm64/tree/master/put-version-file-here)` folders.

duncan999007, 2022-03-07:
I agree with xF4m3, it sounds like it's an issue with the Ubuntu version. 

I can confirm that I have more than enough storage space on those volumes when running the container, and I've followed those directions for building exactly. 

I'll try on 18.04 if there's not a recommended/confirmed version

xF4m3, 2022-03-07:
Very weird, i also got 200+GB of space, so that should not be an issue. I do not update anything in the docker at all.
Fresh install of Ubuntu 18.04, just installed docker and pasted the command from the readme. The Container starts but even waiting 1 hour does not change anything, postgres and ulp-go-app seem to start and have ports, but thats it. With debian 11 for example, the container starts but nothing in it starts at all, so no postgress and no ulp-go-app either.

I also tried installing everything manually like in the dockerfile, but then i am missing a few config files, not sure whats different when i do it manually.

duncan999007, 2022-03-07:
After setting up 18.04 and re-running with everything the same, I'm getting the UNVR setup screen. Mobile app also picks it up, but won't start setup. 

I can go through the setup process in browser but at the very end it hits me with 'An unexpected error occurred during setup.'
Doesn't do anything when I try to download support file. 

Tried with local and UI.com account.

duncan999007, 2022-03-07:
I seem to be having all the same log outputs and issues as this guy: https://github.com/markdegrootnl/unifi-protect-arm64/issues/6

I tried with 1.17.3 as well and it's giving the same error. 

The errors that stand out: 

/srv/unifi-protect/logs/errors.log
2022-03-07T18:15:32.040Z - error: DEVICE NOT REGISTERED unregistered-hardware

systemctl status unifi-core
Mar 07 13:19:19 UNVR sudo[3478]: unifi-core : unable to resolve host UNVR
Mar 07 13:19:19 UNVR sudo[3478]: unifi-core : problem with defaults entries ; TTY=unknown ; PWD=/usr/share/unifi-core ; USER=root ;
Mar 07 13:19:19 UNVR sudo[3478]: unifi-core : TTY=unknown ; PWD=/usr/share/unifi-core ; USER=root ; COMMAND=/sbin/ubnt-systool sshd

SomKen, 2022-03-17:
> 

I love you. I needed this...

iMonZ, 2022-05-14:
> The Host OS should not have any influence if you're running the Docker container.
>
> You should not run any apt updates inside the docker container, this wil break it for sure.
>
> If the container [...]

Where can I finde the UNVR firmware?

markdegrootnl, 2022-05-14:
> Where can I finde the UNVR firmware?
> 
> Okay I found a firmware that’s a year old but how can I extract this files from the bin?

https://community.ui.com/releases

iMonZ, 2022-05-14:
> > Where can I finde the UNVR firmware?
> > Okay I found a firmware that’s a year old but how can I extract this files from the bin?
> 
> https://community.ui.com/releases

And what should I do with the bin file?

uringlas, 2022-05-31:
> > > Where can I finde the UNVR firmware?
> > > Okay I found a firmware that’s a year old but how can I extract this files from the bin?
> > 
> > 
> > https://community.ui.com/releases
> 
> And what should I do with the bin file?

here is explained what to do with the bin.
https://github.com/markdegrootnl/unifi-protect-arm64/blob/master/doc/Extract_deb_files_from_firmware.md

iMonZ, 2022-05-31:
> here is explained what to do with the bin. https://github.com/markdegrootnl/unifi-protect-arm64/blob/master/doc/Extract_deb_files_from_firmware.md

Thanks this actually is helping!

cstrassmair, 2023-10-10:
Hello
I extracted everything from firmware but got problems in building
Here the errors

The following NEW packages will be installed:
  ubnt-archive-keyring
0 upgraded, 1 newly installed, 0 to remove and 0 not upgraded.
Need to get 0 B/5342 B of archives.
After this operation, 15.4 kB of additional disk space will be used.
Get:1 /ubnt-archive-keyring_1.0.3~17+gaebba8f_arm64.deb ubnt-archive-keyring arm64 1.0.3~17+gaebba8f [5342 B]
debconf: delaying package configuration, since apt-utils is not installed
dpkg-deb: error: archive '/ubnt-archive-keyring_1.0.3~17+gaebba8f_arm64.deb' uses unknown compression for member 'control.tar.zst', giving up
dpkg: error processing archive /ubnt-archive-keyring_1.0.3~17+gaebba8f_arm64.deb (--unpack):
 dpkg-deb --control subprocess returned error exit status 2
Errors were encountered while processing:
 /ubnt-archive-keyring_1.0.3~17+gaebba8f_arm64.deb
E: Sub-process /usr/bin/dpkg returned an error code (1)
The command '/bin/sh -c apt-get -y --no-install-recommends install /ubnt-archive-keyring_*_arm64.deb     && echo 'deb https://apt.artifacts.ui.com bullseye main release beta' > /etc/apt/sources.list.d/ubiquiti.list     && chmod 666 /etc/apt/sources.list.d/ubiquiti.list     && apt-get update     && apt-get -y --no-install-recommends install /*.deb unifi-protect     && rm -f /*.deb     && rm -rf /var/lib/apt/lists/*     && echo "exit 0" > /usr/sbin/policy-rc.d     && sed -i 's/redirectHostname: unifi//' /usr/share/unifi-core/app/config/config.yaml     && mv /sbin/mdadm /sbin/mdadm.orig     && mv /usr/sbin/smartctl /usr/sbin/smartctl.orig     && systemctl enable storage_disk dbpermissions    && pg_dropcluster --stop 9.6 main     && sed -i 's/rm -f/rm -rf/' /sbin/pg-cluster-upgrade     && sed -i 's/OLD_DB_CONFDIR=.*/OLD_DB_CONFDIR=\/etc\/postgresql\/9.6\/main/' /sbin/pg-cluster-upgrade' returned a non-zero code: 100