markdegrootnl/unifi-protect-arm64 #14: docker container started but protect not started no log output

cstrassmair, 2022-04-22:
Hello!
I saw this project here and with the raspberries i have it looks promising with higher Version numbers of protect

I wanted to start this container on a raspberry pi 4 4gb with raspberry os lite 64 bit with 128 GB sdcard and an usb disk connected with 4TB space.
Its mounted as /mnt/usb1 and i created a symlink to /storage.

I started the container with 1:1 as written on the Info Page.
It creates the folders on the usb disk and with docker ps says that unifi-protect-arm64 is started.

But on https://raspberryip there is nothing started and docker logs gives no output.

Am i missing something?

thanks
christoph

vehvis, 2022-08-12:
I am suffering from the same problem trying to run the container on Ubuntu 22.04.1 (arm64) on a RasPi 4B.

When opening a shell in the running container, the only process I see running is systemd. Doing some digging there should be at least postgresql, unifi-core and unifi-protect services up in addition to systemd.

When trying to start services manually they typically fail at "cannot connect to bus".

I'm no systemd expert, but this seems to indicate that there is some problem with the container's use of whatever it requires from the host to be able to run systemd. This scenario (systemd inside docker) is not at all common for, I guess, good reasons...

iglu-sebastian, 2022-09-26:
Adding
```
    cap_add:
      - dac_read_search
      - sys_admin
    security_opt:
      - apparmor=unconfined
 ```
 Even without `privileged: true` seems to do the trick for me

xorinzor, 2022-11-03:
On my Raspberry Pi I had to mount the following volumes as well:

```
- /run/dbus:/run/dbus
- /run/systemd:/run/systemd
```

Once I added the kernel parameter to fix the issue as described by the README it took a couple of minutes before unifi protect was listening on the web ports. Interestingly enough there still was no output whatsoever, the only difference was that when looking at the processes in the container I could see the postgresql and unifi services running.

cstrassmair, 2022-11-13:
So has anyone got it running? I would love to get it running.
Can anyone guide me?
Thanks Christoph

Qonstrukt, 2022-11-29:
I have the exact same issue. Other containers are running without issue. Really weird that this image just gives no output at all.

helmut72, 2023-04-15:
Same for me. No log output after starting the container. In the container only systemd is running.

I'm using Ubuntu 22.04 as the host for my other systems. Also tried RPI OS (Bullseye), because it's the same Debian version like in the Docker container (look at the Dockerfile). But also doesn't work.

Are there specific requirements for the host OS or Docker daemon.json settings and if yes, would be great to publish it in the README.md

helmut72, 2023-04-15:
Works for me now. I don't read the other logs. Need to boot the system with kernel parameter `systemd.unified_cgroup_hierarchy=0`

Works great since 1.5h

mattk926, 2023-05-18:
> Works for me now. I don't read the other logs. Need to boot the system with kernel parameter `systemd.unified_cgroup_hierarchy=0`
> 
> Works great since 1.5h

Are you running on a Pi? if so where do you enter "systemd.unified_cgroup_hierarchy=0" I've been trying on 22.04 with no luck.

xF4m3, 2023-08-03:
Alright this has already helped me alot but i still have one issue i am unable to resolve, maybe someone else can help me with this. Together with the ```--cgroupns=host``` and removing the ro from ```-v /sys/fs/cgroup:/sys/fs/cgroup```

This command successfully starts the container and works like expected.
```docker run -it --name unifi-protect --cgroupns=host --privileged --tmpfs /run --tmpfs /run/lock --tmpfs /tmp -v /sys/fs/cgroup:/sys/fs/cgroup -v /storage/srv:/srv -v /storage/data:/data -v /storage/persistent:/persistent --network host     -e /dev/nvme0n1=/dev/sda1 markdegroot/unifi-protect-arm64```

But now i want to start it up using docker compose so i converted the cmd into the following docker compose file, but with that i get no log output again.

```services:
  unifi-protect:
    image: markdegroot/unifi-protect-arm64
    cgroup: host
    container_name: unifi-protect
    volumes:
      - /sys/fs/cgroup:/sys/fs/cgroup
      - /storage/srv:/srv
      - /storage/data:/data
      - /storage/persistent:/persistent
    environment:
      - /dev/nvme0n1=/dev/sda1
    restart: always
    tmpfs:
      - /run
      - /run/lock
      - /tmp
    privileged: true
    network_mode: host```
    
    This should be the same, but for some reason its not working and i am at my wits end. Hope someone can help with this.

helmut72, 2023-08-03:
You really don't see any logs with `docker logs -f unifi-protect` command?

xF4m3, 2023-08-03:
Yeah i usually start it with docker compose up, without the -d flag but starting it detached and using ```docker logs -f unifi-protect``` it still does not show anything at all.

helmut72, 2023-08-03:
Do you see logs from other apps, that you start with a docker-compose.yml? Any special (log) configuration in `/etc/docker/daemon.json`?

xF4m3, 2023-08-03:
No special log configuration, its a fresh ubuntu 22.04 orangepi install, with the kernel parameter added, docker-ce and docker-compose-plugin installed and up to date. Then I added the compose file. I could spin up a diffrent container with compose just to test, but i would assume that should work fine.

helmut72, 2023-08-03:
In your docker run, you use `--cgroupns=host`, in your docker-compose it's `cgroup: host` only. Without ns after cgroup.

I was just a test, but I used this one:

```
version: '3'

services:
  unifi-protect:
    container_name: unifi-protect
    #image: markdegroot/unifi-protect-arm64:1.20.3
    image: markdegroot/unifi-protect-arm64:latest
    restart: always 
    privileged: true
    cap_add:
      - dac_read_search
      - sys_admin
    security_opt:
      - apparmor=unconfined
      - seccomp=unconfined
    environment:
      - STORAGE_DISK=/dev/mmcblk0p2
    volumes:
      - /sys/fs/cgroup:/sys/fs/cgroup:ro
      - /storage/srv:/srv
      - /storage/data:/data
      - /storage/persistent:/persistent
    tmpfs:
      - /run
      - /run/lock
      - /tmp
    network_mode: host
```

xF4m3, 2023-08-03:
So from my google search on how to do ```--cgroupns=host``` in a compose file you do ```cgroup: host```
If i try to do ```cgroupns: host``` that does not exist.

Also trying your file also does not work/show any output at all. 

I am thinking it has to do with the cgroup, since thats seems to be the only part that slightly differs from my docker run cmd.

voslucas, 2023-08-23:
> Yeah i usually start it with `docker compose up`, without the `-d` flag but starting it detached and using `docker logs -f unifi-protect` it still does not show anything at all. --> only with the compose file, with the docker run cmd, everything works as intended and i get logs aswell as a working container.
> 
> Edit: oh and also just to add some info, i am also starting the system with `systemd.unified_cgroup_hierarchy=0` atleast it is added to the kernel start parameter. is there a cmd to check its actually doing something?

Yes! Run `docker system info` and look for `Cgroup Version` in the output.
Did you solve the docker compose problem? I've experiencing the same thing, same OS, same Pi4... No logs, nothing..

xXValiXx, 2023-12-16:
Hey anyone got it working? My docker and docker compose is stuck. It does not output logs or anything. 
I am using a raspberry pi 4 with 8GB memory and 500GB external SSD.
I am using aarch64.

xXValiXx, 2023-12-19:
@xF4m3 How did you get it to work? I have the same problem. Can you help me please?

xF4m3, 2024-03-10:
> @xF4m3 How did you get it to work? I have the same problem. Can you help me please?

Never did get it to work with a docker compose file no matter what i tried, something is diffrent with the cgroup via compose and via cmd line, no idea what.

created the docker via: `docker run -it --name unifi-protect --cgroupns=host --privileged --restart=always --tmpfs /run --tmpfs /run/lock --tmpfs /tmp -v /sys/fs/cgroup:/sys/fs/cgroup -v /storage/srv:/srv -v /storage/data:/data -v /storage/persistent:/persistent --network host -e STORAGE_DISK=/dev/nvme0n1p2 markdegroot/unifi-protect-arm64`

and i can start and stop doing `docker start unifi-protect` `docker stop unifi-protect`