markdegrootnl/unifi-protect-arm64 #17: using qemu to run this on x86?

gurkburk76, 2023-04-29:
I don't know enough to get this running under docker on a x86 machine using qemu to emulate a arm64 but i think it would be possible, does anyone have or could make a guide?

kiwimato, 2023-06-04:
After a few days of trial and error I was able to run it on x86, I made a fork and tried to document everything [here](https://github.com/kiwimato/unifi-protect-arm64#running-unifi-protect-arm64-on-x86-machines) 
Feel free to give it a try and let me know if you have any issues or ideas on out to improve it. 
Of course, feel free to contribute!

gurkburk76, 2023-06-07:
I've run into a few issues while trying to get this going, something about the database not getting created from what i can tell from the logs.
Running this:
ubuntu 22.xx LTS (same problem with 20.xx LTS)

cat /proc/cmdline
BOOT_IMAGE=/boot/vmlinuz-5.15.0-73-generic root=UUID=31f93f3c-8482-49b5-88b8-535c3f9a1269 ro splash quiet systemd.unified_cgroup_hierarchy=0 apparmor=0 vt.handoff=7

my config to start the container:
docker run --rm -d --name unifi-protect  \
    --privileged \
    --tmpfs /run \
    --tmpfs /run/lock \
    --tmpfs /tmp \
    -v /sys/fs/cgroup:/sys/fs/cgroup:ro \
    -v /storage/srv:/srv \
    -v /storage/data:/data \
    -v /storage/persistent:/persistent \
    --network host \
    -e STORAGE_DISK=/dev/sda2 \
    markdegroot/unifi-protect-arm64

logs of errors, there are a few more errors but it would make sense that some services won't start if the database dosen't get created ;)
If you need more just tell me what you need.

docker exec -it unifi-protect /bin/bash
----
: pam_unix(sudo:session): session opened for user root(uid=0) by (uid=999)
Jun 05 16:53:21 streams sudo[2336]: pam_unix(sudo:session): session closed for user root
Jun 05 16:53:55 streams systemd[1]: ulp-go.service: start operation timed out. Terminating.
Jun 05 16:53:55 streams systemd[1]: ulp-go.service: Failed with result 'timeout'.
Jun 05 16:53:55 streams systemd[1]: ulp-go.service: Unit process 1633 (ulp-go-app) remains running after unit stopped.
Jun 05 16:53:55 streams systemd[1]: ulp-go.service: Unit process 2486 (sleep) remains running after unit stopped.
Jun 05 16:53:55 streams systemd[1]: Failed to start ULP-GO.
Jun 05 16:53:56 streams start.sh[1612]: waiting for ulp-go app starting completed..............................................................................................
Jun 05 16:54:01 streams systemd[1]: ulp-go.service: Scheduled restart job, restart counter is at 3.
Jun 05 16:54:01 streams systemd[1]: Stopped ULP-GO.
Jun 05 16:54:01 streams systemd[1]: ulp-go.service: Found left-over process 1633 (ulp-go-app) in control group while starting unit. Ignoring.
Jun 05 16:54:01 streams systemd[1]: This usually indicates unclean termination of a previous run, or service implementation deficiencies.
Jun 05 16:54:01 streams systemd[1]: Starting ULP-GO...
Jun 05 16:54:01 streams pre-start.sh[2493]: /bin/cat: /tmp/.restore_status: No such file or directory
Jun 05 16:54:03 streams sudo[2506]:     root : PWD=/ ; USER=postgres ; COMMAND=/usr/bin/psql -tAc SELECT 1 FROM pg_roles WHERE rolname='ulp-go'
Jun 05 16:54:03 streams sudo[2506]: pam_unix(sudo:session): session opened for user postgres(uid=104) by (uid=0)
Jun 05 16:54:05 streams systemd[1]: unifi-core.service: start operation timed out. Terminating.
Jun 05 16:54:05 streams systemd[1]: unifi-core.service: Failed with result 'timeout'.
Jun 05 16:54:05 streams systemd[1]: Failed to start UniFi Core.
Jun 05 16:54:05 streams systemd[1]: Started Unifi Pion Gateway service.
Jun 05 16:54:05 streams systemd[1]: Starting UniFi Protect...
Jun 05 16:54:05 streams unifi-pion-gw-service.sh[2527]: /usr/bin/unifi-pion-gw-service.sh: line 6: /data/unifi-core/logs/unifi-pion-gw-crash.log: No such file or directory
Jun 05 16:54:05 streams systemd[1]: unifi-core.service: Scheduled restart job, restart counter is at 1.
Jun 05 16:54:05 streams systemd[1]: Stopped UniFi Core.
Jun 05 16:54:05 streams unifi-pion-gw-service.sh[2521]: /usr/bin/unifi-pion-gw-service.sh: line 10: /data/unifi-core/logs/unifi-pion-gw-crash.log: No such file or directory
Jun 05 16:54:05 streams su[2529]: (to postgres) root on none
Jun 05 16:54:05 streams su[2529]: pam_unix(su:session): session opened for user postgres(uid=104) by (uid=0)
Jun 05 16:54:05 streams systemd[1]: unifi-pion-gw.service: Succeeded.
Jun 05 16:54:07 streams pre-start[2540]: createuser: error: creation of new role failed: ERROR:  role "unifi-protect" already exists
Jun 05 16:54:07 streams su[2529]: pam_unix(su:session): session closed for user postgres
Jun 05 16:54:08 streams pre-start[2567]: /bin/chown: missing operand after ‘unifi-protect:unifi-protect’
Jun 05 16:54:08 streams pre-start[2567]: Try '/bin/chown --help' for more information.
Jun 05 16:54:08 streams pre-start[2576]: /bin/chown: cannot access '/srv/unifi-protect/cv': No such file or directory
Jun 05 16:54:11 streams systemd[1]: unifi-pion-gw.service: Scheduled restart job, restart counter is at 1.
Jun 05 16:54:11 streams systemd[1]: Stopped Unifi Pion Gateway service.
.
.
.
Jun 05 16:55:51 streams sudo[2995]:     root : PWD=/ ; USER=postgres ; COMMAND=/usr/bin/createuser ulp-go
Jun 05 16:55:51 streams sudo[2995]: pam_unix(sudo:session): session opened for user postgres(uid=104) by (uid=0)
Jun 05 16:55:53 streams pre-start.sh[2998]: createuser: error: creation of new role failed: ERROR:  role "ulp-go" already exists
Jun 05 16:55:53 streams sudo[2995]: pam_unix(sudo:session): session closed for user postgres
Jun 05 16:55:53 streams pre-start.sh[2957]: create user failed
Jun 05 16:55:53 streams pre-start.sh[2957]: ERROR<[1]>: create user ulp-go
[... 27 lines cut ...]
Jun 05 16:56:25 streams node[2938]:     at addChunk (node:internal/streams/readable:315:12)
Jun 05 16:56:25 streams node[2938]:     at readableAddChunk (node:internal/streams/readable:289:9)
Jun 05 16:56:25 streams node[2938]:     at Socket.Readable.push (node:internal/streams/readable:228:10)
[... cut ...]
Jun 05 16:56:25 streams node[2938]:   file: 'postinit.c',
Jun 05 16:56:25 streams node[2938]:   line: '885',
Jun 05 16:56:25 streams node[2938]:   routine: 'InitPostgres'
Jun 05 16:56:25 streams node[2938]: }
Jun 05 16:56:29 streams sudo[3459]:   ulp-go : PWD=/data/ulp-go ; USER=root ; COMMAND=/sbin/ubnt-tools id
Jun 05 16:56:29 streams sudo[3459]: pam_unix(sudo:session): session opened for user root(uid=0) by (uid=999)
Jun 05 16:56:30 streams sudo[3459]: pam_unix(sudo:session): session closed for user root

kiwimato, 2023-06-07:
@gurkburk76 hmm it seems unifi-core and ulp-go services hit the timeout too soon, which means for some reason the timeout I set was ignored

gurkburk76, 2023-06-09:
OK, sooo, what to do? :)

kiwimato, 2023-06-09:
@gurkburk76 waait I think there's a mistake in your command.
you're still running `markdegroot/unifi-protect-arm64` (the end of the command). You just changed the `name` of the container to `unifi-protect`. If you don't want to use docker-compose as the [docs](https://github.com/kiwimato/unifi-protect-arm64#running-unifi-protect-arm64-on-x86-machines) suggested, you will have to build it using this command:
```
docker build --platform linux/arm64  -t unifi-patch -f patch/Dockerfile.patch patch
```

And then modify your command by replacing `markdegroot/unifi-protect-arm64` at the end with `unifi-patch`

Btw I enabled the issues on my forked repository to avoid spamming Mark until we get a working version, so feel free to open an issue there if you still need help: https://github.com/kiwimato/unifi-protect-arm64/issues

kiwimato, 2023-06-12:
TL;DR for [the thread](https://github.com/kiwimato/unifi-protect-arm64/issues/2) mentioned above for future reference:
Running Unifi Protect in an emulated container on x86 is not feasible due to the performance impact caused by the emulator overhead.

Even though increasing some systemd service timeouts can make it work on a fast enough CPU, eventually it still crashed in various ways and I don't have the time or energy to invest into right now. 

Just for future reference a comparison of initial start time between running it on a 2 x 8 core CPU E5-2630 v3 @ 2.40GHz baremetal machine (emulated aarch64 on x86) vs a Raspberry PI 4: 4 min vs 1 min.

closed 2023-06-19 by gurkburk76