markdegrootnl/unifi-protect-arm64 #19: Connecting to UI just hangs, unifi-core restarts every 2 minutes

kiwimato, 2023-06-13:
The problem is that it works initially, on the first setup, I can connect cameras to it, and it works great.

However, if I stop and start the container, when I try to connect to the UI it simply hangs. Dev console says the browser is able to get some files successfully as it can also be seen in the logs, but then it just hangs and I can't figure out what's wrong.

The unifi-core service just restarts every 2 minute indefinitely.

Did any of you guys experience this before?

I already set `systemd.unified_cgroup_hierarchy=0`:
```
root@pi:/home/pi# cat /proc/cmdline 
coherent_pool=1M 8250.nr_uarts=0 snd_bcm2835.enable_compat_alsa=0 snd_bcm2835.enable_hdmi=1  smsc95xx.macaddr=DC:A6:32:FD:31:7A vc_mem.mem_base=0x3ec00000 vc_mem.mem_size=0x40000000  zswap.enabled=1 zswap.zpool=z3fold zswap.compressor=zstd dwc_otg.lpm_enable=0 console=tty1 root=LABEL=writable rootfstype=ext4 rootwait fixrtc quiet splash systemd.unified_cgroup_hierarchy=0
```

Hardware: Raspberry PI 4 B, 8GB mem
OS: Ubuntu 22.04

```
Jun 14 01:30:37 pi node[1308]: 2023-06-13T23:30:37.148Z - [http] : https 705ms 200 GET /
Jun 14 01:30:37 pi sudo[4210]: unifi-core : PWD=/usr/share/unifi-core/app ; USER=root ; COMMAND=/sbin/ubnt-systool network-speed enp0s2
Jun 14 01:30:37 pi sudo[4210]: pam_unix(sudo:session): session opened for user root(uid=0) by (uid=998)
Jun 14 01:30:37 pi sudo[4210]: pam_unix(sudo:session): session closed for user root
Jun 14 01:30:37 pi node[1308]: 2023-06-13T23:30:37.245Z - [system] : error reading network data Error: Unexpected speed=0 received by cmd=/usr/bin/sudo -n /sbin/ubnt-systool network-speed enp0s2
Jun 14 01:30:37 pi node[1308]:     at dl (/usr/share/unifi-core/app/service.js:22:6530)
Jun 14 01:30:37 pi node[1308]:     at async Promise.all (index 1)
Jun 14 01:30:37 pi node[1308]:     at r (/usr/share/unifi-core/app/service.js:24:986)
Jun 14 01:30:37 pi node[1308]:     at async Promise.all (index 1)
Jun 14 01:30:37 pi node[1308]:     at Object.s [as getClientNetwork] (/usr/share/unifi-core/app/service.js:24:1161)
Jun 14 01:30:37 pi node[1308]:     at async Promise.all (index 7)
Jun 14 01:30:37 pi node[1308]:     at $ (/usr/share/unifi-core/app/service.js:83:25082)
Jun 14 01:30:37 pi node[1308]: 2023-06-13T23:30:37.310Z - [http] : https 26ms 200 GET /floorplanDeps.f0b03dfbe9117a75e741.js
Jun 14 01:30:37 pi node[1308]: 2023-06-13T23:30:37.312Z - [http] : https 24ms 200 GET /icons.0ce1c9f05fceb5191a86.js
Jun 14 01:30:37 pi node[1308]: 2023-06-13T23:30:37.313Z - [http] : https 21ms 200 GET /main.0c06c61c.css
Jun 14 01:30:37 pi node[1308]: 2023-06-13T23:30:37.394Z - [http] : https 57ms 200 GET /lottie.674941a7605abf130e9c.js
Jun 14 01:30:37 pi node[1308]: 2023-06-13T23:30:37.396Z - [http] : https 49ms 200 GET /vendor.e36d615c9f4437a5f30d.js
Jun 14 01:30:37 pi node[1308]: 2023-06-13T23:30:37.397Z - [http] : https 46ms 200 GET /main.030796c94c32707d75ef.js
Jun 14 01:31:01 pi systemd[1]: systemd-timedated.service: Succeeded.
░░ Subject: Unit succeeded
░░ Defined-By: systemd
░░ Support: https://www.debian.org/support
░░ 
░░ The unit systemd-timedated.service has successfully entered the 'dead' state.
Jun 14 01:32:07 pi systemd[1]: unifi-core.service: Watchdog timeout (limit 2min)!
Jun 14 01:32:07 pi systemd[1]: unifi-core.service: Killing process 1308 (node) with signal SIGABRT.
Jun 14 01:32:08 pi systemd[1]: unifi-core.service: Main process exited, code=dumped, status=6/ABRT
```

kiwimato, 2023-06-13:
Either that, or it simply crashes rapidly by itself, and not sure if this is a Unifi bug, or something else:
```
Jun 14 01:51:29 pi node[1757]: 2023-06-13T23:51:29.604Z - [app] : Attached system.ustorage middleware 0.007s
Jun 14 01:51:29 pi node[1757]: 2023-06-13T23:51:29.605Z - [app] : Attached system middleware 1.018s
Jun 14 01:51:29 pi node[1757]: 2023-06-13T23:51:29.610Z - [app] : http middleware taking a long time to attach. 1.061s
Jun 14 01:51:29 pi node[1757]: 2023-06-13T23:51:29.684Z - [app] : Attached http middleware 1.135s
Jun 14 01:51:29 pi node[1757]: 2023-06-13T23:51:29.686Z - [http] : HTTP Server is now listening on port 80
Jun 14 01:51:29 pi sudo[2075]: unifi-core : PWD=/usr/share/unifi-core/app ; USER=root ; COMMAND=/sbin/mdadm --detail /dev/md3
Jun 14 01:51:29 pi sudo[2075]: pam_unix(sudo:session): session opened for user root(uid=0) by (uid=998)
Jun 14 01:51:29 pi sudo[2075]: pam_unix(sudo:session): session closed for user root
Jun 14 01:51:29 pi node[1757]: 2023-06-13T23:51:29.816Z - [messageBox] : Subscribe token: UUM
Jun 14 01:51:29 pi node[1757]: 2023-06-13T23:51:29.817Z - [messageBox] : Subscribe token: WAN_TRANSITION
Jun 14 01:51:29 pi node[1757]: 2023-06-13T23:51:29.843Z - [messageBox] : Subscribe token: DEVICE_LIST
Jun 14 01:51:29 pi node[1757]: 2023-06-13T23:51:29.846Z - [messageBox] : Subscribe token: CLIENT_LIST
Jun 14 01:51:29 pi node[1757]: 2023-06-13T23:51:29.883Z - [app] : App Startup Complete. VERSION : 3.0.54
Jun 14 01:51:29 pi sudo[2098]: unifi-core : PWD=/usr/share/unifi-core/app ; USER=root ; COMMAND=/sbin/ubnt-systool network-speed enp0s2
Jun 14 01:51:29 pi sudo[2098]: pam_unix(sudo:session): session opened for user root(uid=0) by (uid=998)
Jun 14 01:51:29 pi sudo[2098]: pam_unix(sudo:session): session closed for user root
Jun 14 01:51:30 pi sudo[2106]: unifi-core : PWD=/usr/share/unifi-core/app ; USER=root ; COMMAND=/sbin/ubnt-systool network-speed enp0s2
Jun 14 01:51:30 pi sudo[2106]: pam_unix(sudo:session): session opened for user root(uid=0) by (uid=998)
Jun 14 01:51:30 pi sudo[2106]: pam_unix(sudo:session): session closed for user root
Jun 14 01:51:30 pi node[1757]: 2023-06-13T23:51:30.073Z - [firmware] : Startup: Retrieved auto-update schedule
Jun 14 01:51:30 pi node[1757]: {
Jun 14 01:51:30 pi node[1757]:   frequency: 'weekly',
Jun 14 01:51:30 pi node[1757]:   includeFirmware: true,
Jun 14 01:51:30 pi node[1757]:   includeControllers: true,
Jun 14 01:51:30 pi node[1757]:   day: 0,
Jun 14 01:51:30 pi node[1757]:   hour: 0
Jun 14 01:51:30 pi node[1757]: }
Jun 14 01:51:30 pi node[1757]: 2023-06-13T23:51:30.118Z - [system] : Setting hostname - pi
Jun 14 01:51:30 pi node[1757]: 2023-06-13T23:51:30.147Z - [system.ustorage] : ustorage config
Jun 14 01:51:30 pi node[1757]: { hotspare: false, raid: 'single' }
Jun 14 01:51:30 pi sudo[2122]: unifi-core : PWD=/usr/share/unifi-core/app ; USER=root ; COMMAND=/sbin/ubnt-systool hostname pi
Jun 14 01:51:30 pi sudo[2122]: pam_unix(sudo:session): session opened for user root(uid=0) by (uid=998)
Jun 14 01:51:30 pi node[1757]: 2023-06-13T23:51:30.177Z - [sdnotify] : sd-notify ready()
Jun 14 01:51:30 pi systemd[1]: Started UniFi Core.
░░ Subject: A start job for unit unifi-core.service has finished successfully
░░ Defined-By: systemd
░░ Support: https://www.debian.org/support
░░ 
░░ A start job for unit unifi-core.service has finished successfully.
░░ 
░░ The job identifier is 111.
Jun 14 01:51:30 pi node[1757]: 2023-06-13T23:51:30.182Z - [sdnotify] : Starting Watchdog mode (interval = 120000ms)
Jun 14 01:51:30 pi sudo[2122]: pam_unix(sudo:session): session closed for user root
Jun 14 01:51:30 pi node[1757]: 2023-06-13T23:51:30.211Z - [discovery.responder] : Discovery responder listening on 0.0.0.0:10001
Jun 14 01:51:30 pi sudo[2129]: unifi-core : PWD=/usr/share/unifi-core/app ; USER=root ; COMMAND=/sbin/ubnt-systool cpuload
Jun 14 01:51:30 pi sudo[2129]: pam_unix(sudo:session): session opened for user root(uid=0) by (uid=998)
Jun 14 01:51:30 pi sudo[2131]: unifi-core : PWD=/usr/share/unifi-core/app ; USER=root ; COMMAND=/sbin/ubnt-systool cputemp
Jun 14 01:51:30 pi sudo[2131]: pam_unix(sudo:session): session opened for user root(uid=0) by (uid=998)
Jun 14 01:51:30 pi sudo[2131]: pam_unix(sudo:session): session closed for user root
Jun 14 01:51:30 pi sudo[2140]: unifi-core : PWD=/usr/share/unifi-core/app ; USER=root ; COMMAND=/sbin/ubnt-systool sshd
Jun 14 01:51:30 pi sudo[2140]: pam_unix(sudo:session): session opened for user root(uid=0) by (uid=998)
Jun 14 01:51:30 pi sudo[2146]: unifi-core : PWD=/usr/share/unifi-core/app ; USER=root ; COMMAND=/sbin/ubnt-systool cpuload
Jun 14 01:51:30 pi sudo[2146]: pam_unix(sudo:session): session opened for user root(uid=0) by (uid=998)
Jun 14 01:51:30 pi sudo[2140]: pam_unix(sudo:session): session closed for user root
Jun 14 01:51:30 pi sudo[2152]: unifi-core : PWD=/usr/share/unifi-core/app ; USER=root ; COMMAND=/sbin/ubnt-systool cputemp
Jun 14 01:51:30 pi sudo[2152]: pam_unix(sudo:session): session opened for user root(uid=0) by (uid=998)
Jun 14 01:51:30 pi node[1757]: 2023-06-13T23:51:30.437Z - [firmware] : Running update schedule:
Jun 14 01:51:30 pi node[1757]: {
Jun 14 01:51:30 pi node[1757]:   frequency: 'weekly',
Jun 14 01:51:30 pi node[1757]:   includeFirmware: true,
Jun 14 01:51:30 pi node[1757]:   includeControllers: true,
Jun 14 01:51:30 pi node[1757]:   day: 0,
Jun 14 01:51:30 pi node[1757]:   hour: 0
Jun 14 01:51:30 pi node[1757]: }
[... cut ...]
Jun 14 01:51:34 pi systemd[1]: unifi-core.service: Found left-over process 2051 (sudo) in control group while starting unit. Ignoring.
Jun 14 01:51:34 pi systemd[1]: This usually indicates unclean termination of a previous run, or service implementation deficiencies.
Jun 14 01:51:34 pi systemd[1]: unifi-core.service: Found left-over process 2052 (systemctl) in control group while starting unit. Ignoring.
Jun 14 01:51:34 pi systemd[1]: This usually indicates unclean termination of a previous run, or service implementation deficiencies.
```

bkchaw, 2023-06-14:
make sure you only put the 4 deb files into the folder put-deb-files-here

bkchaw, 2023-06-14:
do you compiled the image yourself? if not do not update protect from within the UI

jorisleem, 2023-06-14:
I have the same problem as Kiwimato.

bkchaw, 2023-06-14:
I also have this problem when I run using docker compose up command but if use docker run -d command then it run without any problems. You can try that

kiwimato, 2023-06-14:
ok, that's super interesting and weird at the same time, I will give it a try.
update on my side is that I managed to build an upgraded image, but it hits the same problem, so I think it's something missing in either volumes or config which is not saved across reboots, but I may be wrong.

bkchaw, 2023-06-14:
You will need this file /lib/systemd/system/unifi-core.serviced.d/cap.conf to fix the permission issues in setting timezone and backup etc in the ui settings

jorisleem, 2023-06-14:
> I also have this problem when I run using docker compose up command but if use docker run -d command then it run without any problems. You can try that

This doesn't work for me. Same result.

bkchaw, 2023-06-14:
> > I also have this problem when I run using docker compose up command but if use docker run -d command then it run without any problems. You can try that
> 
> This doesn't work for me. Same result.

Did you clear the /storage first before starting it?

kiwimato, 2023-06-14:
@jorisleem I figured it out, at least for the newer Protect version(UNVR 3.1.9), the problem was that the Postgres "main" database was not saved properly across container restarts.
I have a working version, if you want to try it out, use this image instead: `ghcr.io/kiwimato/unifi-protect-arm64:dev`

Just please be mindful that it will update your cameras if you adopt them to the latest version automatically, just in case you don't want that since downgrading can be cumbersome.

There's still some testing to be done, and I want to fix the fact that the "main" Postgres DB config is not properly linked in /srv, it's still in /etc/postgresql/main, but at least this works for now until we fix that part.

The changes I made can be found [here](https://github.com/kiwimato/unifi-protect-arm64/pull/3/files) which of course excludes the Unifi debs due to copyright and stuff.

bkchaw, 2023-06-15:
@kiwimato i can't get the storage page to show when using your build.

gurkburk76, 2023-06-15:
So i got my rock64 (4 gigs of mem) sort of working but having a few strange lines in the logs, i haven't gotten to the adop part yet, so i'm sure i'll have even more questions further along.

This might be related to the topic as stuff dosen't quite work and it refuses to save some of my settings.

```
Jun 15 06:56:36 UNVR sudo[19764]: unifi-core : unable to resolve host UNVR: Name or service not known
Jun 15 06:56:36 UNVR sudo[19764]: unifi-core : problem with defaults entries ; PWD=/usr/share/unifi-core/app ; USER=root ;

```
Also, just to be really clear about this, if i pull the image, witch i did, then i don't have to put any deb files anywhere, correct ?

I'm thinking of the [put-deb-files-here](https://github.com/markdegrootnl/unifi-protect-arm64/tree/master/put-deb-files-here) and [put-version-file-here](https://github.com/markdegrootnl/unifi-protect-arm64/tree/master/put-version-file-here) folders.

kiwimato, 2023-06-15:
@bkchaw I have the same problem with storage, I'll look into it. I just pushed a new image which includes `unifi-assets-unvr_2.0.0_arm64.deb`

@gurkburk76 I solved that by naming the UNVR as the host hostname, in your case is DietPi. You could also modify the host /etc/hosts and add: `127.0.0.1 UNVR`  which should make it resolve on the container also.

I do not see `problem with defaults entries ; PWD=/usr/share/unifi-core/app ; USER=root ;` error on my container, not sure what it caused it.

> Also, just to be really clear about this, if i pull the image, witch i did, then i don't have to put any deb files anywhere, correct ?
Correct. The debs are already builtin the image, you don't have to worry about that, but if you want to build your own image, you can, I tried to make it more obvious how that can be done. 

Do you need Bluetooth on the UNVR for that camera? Alternatively you could also use a USB bluetooth adapter, that might work?

gurkburk76, 2023-06-15:
Yeah that camera needs to be adapted with bluetooth, it's the only way iirc.
Yeah i'll fiddle with a usb adapter i have, might work, if it dosen't i could try me rasp 3b+ i think, assuming this docker container "knows" how to use a usb bluetooth. Might work better with the Pi, we'll see what happens.

bkchaw, 2023-06-15:
> Yeah that camera needs to be adapted with bluetooth, it's the only way iirc.
> Yeah i'll fiddle with a usb adapter i have, might work, if it dosen't i could try me rasp 3b+ i think, assuming this [...]

@gurkburk76 you can use your handphone protect app to adopt. I adopted mine this way.

bkchaw, 2023-06-15:
> @bkchaw I have the same problem with storage, I'll look into it.  

that's cool! while you are at that maybe you can figure out how to get unifi-os to show the real hdd info instead of some fake info?

kiwimato, 2023-06-15:
@bkchaw the script should already show real data as long as you set STORAGE_DISK env var.  It's a bit hard to reverse engineer why storage page doesn't work, I checked everything I could, but every seems fine so far, can't find anything in the logs

```
root@pi:/var/log/postgresql# ustorage disk
[
    {
        "action": "none",
        "ata": "ATA8-ACS",
        "bad_sector": 0,
        "estimate": null,
        "firmware": "AX001Q",
        "healthy": "good",
        "life_span": null,
        "model": "UniFi Protect fake disk",
        "poweronhrs": 1,
        "progress": null,
        "reason": null,
        "rpm": 5400,
        "sata": "SATA 2.6",
        "serial": "X0JNP396T",
        "size": 9920821608448,
        "slot": 1,
        "state": "normal",
        "temperature": 49,
        "threshold": 10,
        "type": "HDD"
    }
]
root@pi:/var/log/postgresql# ustorage space
[
    {
        "action": "none",
        "device": "sdb1",
        "estimate": 0,
        "health": "health",
        "progress": null,
        "reasons": [],
        "space_type": "primary",
        "total_bytes": 9920821608448,
        "used_bytes": 2418188288
    }
]
root@pi:/var/log/postgresql# cat `which ustorage`
#!/bin/bash

source /etc/default/storage_disk
disk="${STORAGE_DISK:-/dev/sda1}"
device=$(basename ${disk})
sbytes=$(df -B1 --output=size ${disk} | awk 'NR==2 {print $1}')
sused=$(df -B1 --output=used ${disk} | awk 'NR==2 {print $1}')

case $1 in
    disk)
	cat <<EOT
[
    {
        "action": "none",
        "ata": "ATA8-ACS",
        "bad_sector": 0,
        "estimate": null,
        "firmware": "AX001Q",
        "healthy": "good",
        "life_span": null,
        "model": "UniFi Protect fake disk",
        "poweronhrs": 1,
        "progress": null,
        "reason": null,
        "rpm": 5400,
        "sata": "SATA 2.6",
        "serial": "X0JNP396T",
        "size": ${sbytes},
        "slot": 1,
        "state": "normal",
        "temperature": 49,
        "threshold": 10,
        "type": "HDD"
    }
]
EOT
	;;
    space)
	cat <<EOT
[
    {
        "action": "none",
        "device": "${device}",
        "estimate": 0,
        "health": "health",
        "progress": null,
        "reasons": [],
        "space_type": "primary",
        "total_bytes": ${sbytes},
        "used_bytes": ${sused}
    }
]
EOT
	;;
 config)
	cat <<EOT
{
    "hotspare": false,
    "raid": "single"
}
EOT

esac
root@pi:/var/log/postgresql# 
```

gurkburk76, 2023-06-15:
Ok cool, i got it adopted via the app, now i just have to find where the password is stored for the camera if i want to access it via http/https. it's supposed to be in protect/general/something something but i haven't found it yet.

Also, is updating unifi os + protect supposed to work if done via the webpage os do i have to rebuild the image with the right pakages? I tried to update via the gui and it's been updating for a while now so it seems a bit fishy :)

bkchaw, 2023-06-15:
@gurkburk76  you can’t update via the gui , it will break stuffs. You got to rebuild the image

jorisleem, 2023-06-15:
> @jorisleem I figured it out, at least for the newer Protect version(UNVR 3.1.9), the problem was that the Postgres "main" database was not saved properly across container restarts. That and also for the newer version "main" db is also upgraded to PostgreSQL 14 and some auth mechanisms were enforced to use passwords instead of nothing for localhost (which seems to be the Unifi default) I have a working version, if you want to try it out, use this image instead: `ghcr.io/kiwimato/unifi-protect-arm64:dev`
> 
> Just please be mindful that it will update your cameras if you adopt them to the latest version automatically, just in case you don't want that since downgrading can be cumbersome.
> 
> There's still some testing to be done, and I want to fix the fact that the "main" Postgres DB config is not properly linked in /srv, it's still in /etc/postgresql/main, but at least this works for now until we fix that part.
> 
> The changes I made can be found [here](https://github.com/kiwimato/unifi-protect-arm64/pull/3/files) which of course excludes the Unifi debs due to copyright and stuff.

I used your changed to build a new image. It looks like that it works. Need to monitor it for the next few days.

kiwimato, 2023-06-16:
Guys I will give up trying to figure out why the "Storage" page doesn't work for now. It seems to be a Unifi bug and more annoyingly it's simply not showing any errors anywhere, it just hangs. I found other people complaining about the same issue here, which makes me hope Unifi will solve it eventually: https://community.ui.com/questions/Storage-not-loading/1ddb1c1f-f9e7-47d4-a97f-8efe1a46926a

Just to avoid spamming Mark, feel free to open new issues you find with this image here in my fork: https://github.com/kiwimato/unifi-protect-arm64/issues

I'll give this a few weeks of testing and then I'll create a PR towards this repo.

bkchaw, 2023-06-19:
> Guys I will give up trying to figure out why the "Storage" page doesn't work for now. It seems to be a Unifi bug and more annoyingly it's simply not showing any errors anywhere, it just hangs. I [...]

@kiwimato  I think this is the error that prevented storage page from showing...

2023-06-19T07:35:51.654Z - error: [unvrFlashStorageMigration] Failed to check rwfs migration info via ustorage: [
  {
    "code": "invalid_type",
    "expected": "string",
    "received": "undefined",
    "path": [
      "migratable",
      "reason"
    ],
    "message": "Required"
  }
]

travis-south, 2023-07-19:
> There's still some testing to be done, and I want to fix the fact that the "main" Postgres DB config is not properly linked in /srv, it's still in /etc/postgresql/main, but at least this works for now until we fix that part.

@kiwimato , I experiencing the same issue when restart container. Will mapping `/etc/postgresql/main` as volume in docker-compose solve the issue?