markdegrootnl/unifi-protect-arm64 #23: Protect only sees a 67.1MB disk

miguemely, 2023-09-26:
Hi! 

Using 
```
docker run -d --name unifi-protect  \
    --privileged \
    --tmpfs /run \
    --tmpfs /run/lock \
    --tmpfs /tmp \
    -v /sys/fs/cgroup:/sys/fs/cgroup:ro \
    -v /storage/srv:/srv \
    -v /storage/data:/data \
    -v /storage/persistent:/persistent \
    --network host \
    -e STORAGE_DISK=/dev/nvme0n1p1 \
    markdegroot/unifi-protect-arm64
```

I notice the Protect only sees a 67.1MB Disk. 
[image]

Could it be using /dev? 

```
root@nvr:/# docker exec -it unifi-protect sh
# df -h
Filesystem      Size  Used Avail Use% Mounted on
overlay         116G  5.1G  106G   5% /
tmpfs            64M     0   64M   0% /dev
shm              64M   56K   64M   1% /dev/shm
/dev/nvme0n1p1  916G  176M  870G   1% /srv
tmpfs           3.9G   17M  3.8G   1% /run
tmpfs           3.9G  5.5M  3.9G   1% /tmp
/dev/mmcblk0p2  116G  5.1G  106G   5% /etc/hosts
tmpfs           3.9G     0  3.9G   0% /run/lock
tmpfs           4.0M     0  4.0M   0% /sys/fs/cgroup
tmpfs           256M     0  256M   0% /var/opt/unifi-protect/tmp
#
```

AnthonyOGorman, 2023-10-16:
Did you ever figure this out?
Mine was working fine but I decided to build the latest OS (3.1.16) and now it says there is no disk but happily records to it fine.

belangerpascal, 2023-11-01:
Hi Guys, I'm having somewhat of the same issue. I just did a fresh install, got cgroup stuff figured out and protect is up and running and I was even able to adopt a G4 Instant. The issue I have is that protect or unifi core has no storage showing up. If I run mdadm from the container, it show's the proper disk space and the volume mounts looks fine.

wgrando1993, 2024-06-26:
HI Guys, i found the solve this problem, im using a ssd nvme on my orangepi 5 pro, but, all times show me 67mb, is because the file /etc/storage_disk isnt present on imagem, but, to fix this without re-build, just add these lines to your docker compose file:
`- '/storage/storage_disk:/etc/default/storage_disk'` on your volumes and create a file  storage_disk on /storage directory, put into the fles the following content:

`STORAGE_DISK=/dev/nvme0n1p1`

or the disk you use, in my case it is an nvme

miguemely, 2024-06-26:
Hey @wgrando1993 thanks for that!

So basically,

```
docker run -d --name unifi-protect  \
    --privileged \
    --tmpfs /run \
    --tmpfs /run/lock \
    --tmpfs /tmp \
    -v /sys/fs/cgroup:/sys/fs/cgroup:ro \
    -v /storage/srv:/srv \
    -v /storage/data:/data \
    -v /storage/persistent:/persistent \
    -v /storage/storage_disk:/etc/default/storage_disk \
    --network host \
    -e STORAGE_DISK=/dev/nvme0n1p1 \
    markdegroot/unifi-protect-arm64
```