markdegrootnl/unifi-protect-arm64 #25: Anyone made it to adopt ubnt g4 instant camera (unv-g4-ins) to  unifi-protect-arm64?

breaksona, 2023-09-28:
Pulled the latest image (markdegroot/unifi-protect-arm64:latest), created container and got it running smoothly.

But when I tried to adopt a brand new ubnt g4 instant camera (https://store.ui.com/us/en/products/camera-g4-instant), powered on and heard "welcome to... ready to adopt...", opened Protect APP on my iPhone, got prompt to add G4 instant, scrolled to select my WIFI ssid and input password, then Protect APP showed the radar-scanning animation, without error notice the animation stopped and no camera added.

Tried to do it for several times, WIFI ssid and password was input right, still failed to adopt ubnt g4 instant camera.

Is there any log File to find out what happened?

wagnst, 2023-10-04:
I have a somewhat similar problem, my adoption keeps stuck in adopting state. But the setup itself worked via the app and then I could see the cam in unifi network with an assigned IP.
I set the protect url manually via the cams web interface to adopt it, but this is where it's somehow stuck in adopting state.
The cams logs itself seem to be in /var/log/ but I can see here that it basically is all good.

The ports towards my protect are all good, I've tested them with a telnet from the cam (via ssh).

Not sure it helps you to continue debugging your issue.

breaksona, 2023-10-07:
Hello Mr Wagner,

Do you use g4 instant camera?

I can't see the cam in unifi protect app or web console. The setup didn't
work via the app or unifi-protect web console.

How did you "set the protect url manually via the cams web interface to
adopt it", I can see my cam was assigned an ip in my Router and ping to the
cam is ok.

Regards,
Kane

freechelmi, 2023-10-25:
>  I can see my cam was assigned an ip in my Router and ping to the cam is ok. Regards, Kane

Then go to the web interface of the cam , log with ubnt/ubnt and set the protect URL