markdegrootnl/unifi-protect-arm64 #26: Error while building docker

xXValiXx, 2023-12-19:
Files are filled as mentioned in docs.

Error: dpkg-deb: error: archive '/ubnt-archive-keyring_1.0-1_arm64.deb' uses unknown compression for member 'control.tar.zst', giving up

Docker build fails

waterfallm, 2024-01-19:
I don't get  this error when trying to build v 3.2.9 but i did have to bump nginx to 1.24.0 and then i hit issues with this line

`&& sed -i 's/redirectHostname: unifi//' /usr/share/unifi-core/app/config/config.yaml \`

Commenting out allows the image to build

madmad77-coder, 2024-02-10:
Hello, I also have a problem with the buil and Nginx, how should I get 1.24 into the build??
I have already hidden the line.

     && apt-get -y --no-install-recommends install /*.deb unifi-protect \
     && rm -f /*.deb \
     && rm -rf /var/lib/apt/lists/* \
     && echo "exit 0" > /usr/sbin/policy-rc.d \
    # && sed -i 's/redirectHostname: unifi//' /usr/share/unifi-core/app/config/config.yaml \
     && mv /sbin/mdadm /sbin/mdadm.orig \
     && mv /usr/sbin/smartctl /usr/sbin/smartctl.orig \
     && systemctl enable storage_disk dbpermissions\
     && pg_dropcluster --stop 9.6 main \

but there are always errors when creating.

   66 | >>> && sed -i 's/OLD_DB_CONFDIR=.*/OLD_DB_CONFDIR=\/etc\/postgresql\/9.6\/main/' /sbin/pg-cluster-upgrade
   67 |
--------------------
ERROR: failed to solve: process "/bin/sh -c apt-get -y --no-install-recommends install /ubnt-archive-keyring_*_arm64.deb && echo 'deb https://apt.artifacts.ui .com bullseye main release beta' > /etc/apt/sources.list.d/ubiquiti.list && chmod 666 /etc/apt/sources.list.d/ubiquiti.list && apt-get update && apt-get -y --no-install-recommends install /*.deb unifi-protect && rm -f /*.deb && rm -rf /var/lib/apt/lists/* && echo \"exit 0\" > /usr/sbin /policy-rc.d && mv /sbin/mdadm /sbin/mdadm.orig && mv /usr/sbin/smartctl /usr/sbin/smartctl.orig && systemctl enable storage_disk dbpermissions && pg_dropcluster --stop 9.6 main && sed -i 's/rm -f/rm -rf/' /sbin/pg-cluster-upgrade && sed -i 's/OLD_DB_CONFDIR=.*/OLD_DB_CONFDIR=\\/etc\\/postgresql\\/9.6\\/main /' /sbin/pg-cluster-upgrade" did not complete successfully: exit code: 100
root@RASPBERRY-2:/opt/build-unvr/unifi-protect-arm64# unifi-core : Depends: nginx (>= 1.24.0)

Maybe someone can help me
Thank you very much
​