markdegrootnl/unifi-protect-arm64 #31: Working latest version

dciancu, 2024-05-22:
First of all, massive thank you to @markdegrootnl for the initial project and to @kiwimato, @snowsnoot and @Top-Cat for their forks. 👏 

I have been able to find the time to put together a working repo for the latest version of UNVR firmware, with separate automatic builds for firmware packaged unifi-protect version and latest version.
You can find my project here: https://github.com/dciancu/unifi-protect-unvr-docker-arm64

I am just leaving this here for anyone in need of a more recent version.

All the best to everyone!

gigatux, 2025-01-26:
^ To confirm, everybody should be looking at the above ticket. It works wonderfully and contains very recent versions.

Nice one, @dciancu

dciancu, 2025-02-01:
@gigatux Thank you!