markdegrootnl/unifi-protect-arm64 #32: Unifi Protect can't capture the available storage #9

gnbon, 2024-08-13:
I appreciate the great work on this project. While setting it up, I encountered an issue where my storage was not recognized at all. I'm currently troubleshooting this problem. I of course changed the hostname and added boot commands as well. Either if I have the wrong settings or there are any known issues related to this, please let me know.

### Environments
- Raspberry Pi 4B(Linux raspberrypi 6.6.31)
- Using external 1TB HDD(/dev/sda1)

### df -h output
- host
```
[~]$ df -h
Filesystem      Size  Used Avail Use% Mounted on
udev            1.6G     0  1.6G   0% /dev
tmpfs           380M  1.4M  378M   1% /run
/dev/mmcblk0p2   15G  6.8G  6.6G  51% /
tmpfs           1.9G  148K  1.9G   1% /dev/shm
tmpfs           5.0M   16K  5.0M   1% /run/lock
tmpfs           4.0M     0  4.0M   0% /sys/fs/cgroup
/dev/mmcblk0p1  510M   75M  436M  15% /boot/firmware
/dev/sda1       916G  153M  870G   1% /mnt/storage
tmpfs           380M   40K  380M   1% /run/user/1000
```

- docker
```
# df -h
Filesystem      Size  Used Avail Use% Mounted on
overlay          15G  6.8G  6.6G  51% /
tmpfs            64M     0   64M   0% /dev
shm              64M   56K   64M   1% /dev/shm
tmpfs           1.9G  8.3M  1.9G   1% /run
tmpfs           1.9G  1.1M  1.9G   1% /tmp
/dev/sda1       916G  152M  870G   1% /srv
tmpfs           1.9G     0  1.9G   0% /run/lock
/dev/mmcblk0p2   15G  6.8G  6.6G  51% /etc/hosts
tmpfs           4.0M     0  4.0M   0% /sys/fs/cgroup
tmpfs           1.0G     0  1.0G   0% /var/opt/unifi-protect/tmp
```

### docker-compose.yml
```
services:
  unifi-protect:
    image: dciancu/unifi-protect-unvr-docker-arm64:stable
    tty: true
    hostname: UNVR
    extra_hosts:
      - "UNVR:127.0.1.1"
    container_name: unifi-protect
    stop_grace_period: 2m
    cgroup: host
    volumes:
      - /sys/fs/cgroup:/sys/fs/cgroup
      - /mnt/storage/srv:/srv
      - /mnt/storage/data:/data
      - /mnt/storage/persistent:/persistent
    environment:
      - container=docker
      - STORAGE_DISK=/dev/mmcblk0p2
      - TZ=UTC
    restart: unless-stopped
    cap_add:
      - dac_read_search
      - sys_admin
    security_opt:
      - apparmor=unconfined
      - seccomp=unconfined
    tmpfs:
      - /run
      - /run/lock
      - /tmp
      - /var/run
      - /var/run/lock
    privileged: true
    network_mode: host
```
I've also attached two screenshots showing additional information. Please let me know if you need any further details or clarification on this issue.
[image]
[image]

Zweeny, 2024-10-21:
Hello everyone. 
First of all, thank you very much for the great work on the project. I have been working with it for a while as a 'user' ;-). Currently, the latest 'Edge' version is running on a RockPro64 in Docker (Debian 12, Docker 2.29.2) with a PCI adapter and an SSD.

Everything is running well so far. However, I have the same problem as @gnbon. Unfortunately, the storage is not recognized. I have tried different entries in Docker, but unfortunately without success. (Note: Yes, I have also tried the 'Stable' version.)

Below is my configuration

>Docker:
version: '3'
services:
  unifi-protect:
    image: dciancu/unifi-protect-unvr-docker-arm64:edge
    tty: true
    extra_hosts:
      - "UNVR:127.0.1.1"
    container_name: unifi-protect2
    cgroup: host
    stop_grace_period: 2m
    volumes:
      - /sys/fs/cgroup:/sys/fs/cgroup
      - /ssd/protect2/srv:/srv
      - /ssd/protect2/data:/data
      - /ssd/protect2/persistent:/persistent
    environment:
      - container=docker
      - STORAGE_DISK=/dev/nvme0n1p1
      - TZ=Europe/Berlin
    restart: unless-stopped
    deploy:
      resources:
        limits:
          cpus: '4'
    tmpfs:
      - /run
      - /run/lock
      - /tmp
      - /var/run
      - /var/run/lock
    privileged: true
    cap_add:
      - dac_read_search
      - sys_admin
    security_opt:
      - apparmor=unconfined
      - seccomp=unconfined
    network_mode: host

>VM-CLI
root@UNVR:/# df -h
Filesystem      Size  Used Avail Use% Mounted on
overlay         1.8T  481G  1.3T  28% /
tmpfs            64M     0   64M   0% /dev
shm              64M   56K   64M   1% /dev/shm
tmpfs           1.9G   17M  1.9G   1% /run
tmpfs           1.9G  1.1M  1.9G   1% /tmp
/dev/nvme0n1p1  1.8T  481G  1.3T  28% /srv
tmpfs           1.9G     0  1.9G   0% /run/lock
tmpfs           1.0G  9.1M 1015M   1% /var/opt/unifi-protect/tmp
root@UNVR:/# 

>Docker Host:
rock64@rockpro2:~$ sudo df -h
[sudo] Passwort für rock64:
Dateisystem    Größe Benutzt Verf. Verw% Eingehängt auf
udev            1,9G       0  1,9G    0% /dev
tmpfs           387M    1,6M  385M    1% /run
/dev/mmcblk2p2   56G     18G   36G   33% /
tmpfs           1,9G       0  1,9G    0% /dev/shm
tmpfs           5,0M       0  5,0M    0% /run/lock
/dev/nvme0n1p1  1,8T    480G  1,3T   28% /ssd
/dev/mmcblk2p1  455M    152M  279M   36% /boot

[image]

[image]

Regardless of the errors in the above display, 'events' are being recorded

Anyone have an idea?

f4llenz, 2025-01-11:
Did you figure this out?

gnbon, 2025-01-12:
I turned off interest because I bought the real device. 
How about try @dciancu's fork? 
- https://github.com/dciancu/unifi-protect-unvr-docker-arm64

closed 2025-01-12 by gnbon

dciancu, 2025-01-15:
@gnbon Thank you for pointing to my fork.
If you have a real UNVR, would you mind helping a bit to keep up the project by sharing the output of 2 commands at issue https://github.com/dciancu/unifi-protect-unvr-docker-arm64/issues/23
Your help would be greatly appreciated, thank you.

gnbon, 2025-01-18:
> [@gnbon](https://github.com/gnbon) Thank you for pointing to my fork. If you have a real UNVR, would you mind helping a bit to keep up the project by sharing the output of 2 commands at issue [dciancu/unifi-protect-unvr-docker-arm64#23](https://github.com/dciancu/unifi-protect-unvr-docker-arm64/issues/23) Your help would be greatly appreciated, thank you.

@dciancu Sorry for the late reply, but I need to mention that my device is not a UNVR but rather a Unifi AP that supports Unifi Protect. Since the hardware is different, my output wouldn't be helpful for your investigation.

dciancu, 2025-01-18:
@gnbon Thank you, I agree. I have managed to solve the issued in the end.